ansys/pysherlock #272: update_parts_list() API returning 0 parts updated

aklara-ansys, 2024-04-18:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

This appears to only happen with the pySherlock implementation of update_parts_list(). The java API appears to update the parts as expected. I attached the pySherlock script to use in this example:

import os
from ansys.sherlock.core.launcher import launch_sherlock
from ansys.sherlock.core.types.parts_types import PartsListSearchMatchingMode, PartsListSearchDuplicationMode

sherlock = launch_sherlock()

ANSYS_ROOT = os.getenv("AWP_ROOT242")

sherlock.project.import_odb_archive(
        ANSYS_ROOT + "\\sherlock\\tutorial\\ODB++ Tutorial.tgz",
        True,
        True,
        True,
        True,
        project="Stackup_Tutorial",
        cca_name="Main Board"
        )

#
sherlock.parts.update_parts_list(
    "Stackup_Tutorial",
    "Main Board",
    "Sherlock Part Library",
    PartsListSearchMatchingMode.PART,
    PartsListSearchDuplicationMode.FIRST,
)

---> INFO - parts - update_parts_list - 0 part(s) were updated in the Parts list

But when this is executed on the project in the UI, it works fine and 221 parts get updated:

### 📝 Steps to reproduce

import os
from ansys.sherlock.core.launcher import launch_sherlock
from ansys.sherlock.core.types.parts_types import PartsListSearchMatchingMode, PartsListSearchDuplicationMode

sherlock = launch_sherlock()

try:
    sherlock.project.delete_project("Stackup_Tutorial")
except:
    print("The specified project does not exist.  Proceed normally.")

ANSYS_ROOT = os.getenv("AWP_ROOT242")

sherlock.project.import_odb_archive(
        ANSYS_ROOT + "\\sherlock\\tutorial\\ODB++ Tutorial.tgz",
        True,
        True,
        True,
        True,
        project="Stackup_Tutorial",
        cca_name="Main Board"
        )

#
sherlock.parts.update_parts_list(
    "Stackup_Tutorial",
    "Main Board",
    "Sherlock Part Library",
    PartsListSearchMatchingMode.PART,
    PartsListSearchDuplicationMode.FIRST,
)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

24.2 

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
python -m pip install ansys-sherlock-core
```

aklara-ansys, 2024-04-22:
Yes the output is correct, no parts get updated when executed.

ansnfernand, 2024-06-05:
It's not a PySherlock bug.

closed 2024-06-05 by ansnfernand