ansys/pysherlock #342: launcher.py - launch_sherlock() needs updated for the Linux install path

aklara-ansys, 2024-07-25:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

The implementation of launch_sherlock() in launcher.py references the environment variable AWP_ROOT251. This is not set by the installer on Linux. There needs to be a different implementation to grab the install directory path. No environment variables are set for the installer like they are on Windows, so the implementation must grab that directory in different way. A workaround is to set the environment up yourself "set AWP_ROOT251=/ansys/reg/ansys_inc/v251" for instance.

def _get_base_ansys():
    supported_installed_versions = {
        env_key: path
        for env_key, path in os.environ.items()
        **if env_key.startswith("AWP_ROOT") and os.path.isdir(path)**
    }

    for key in sorted(supported_installed_versions, reverse=True):
        ansys_version = _get_ansys_version_from_awp_root(key)
        if ansys_version >= EARLIEST_SUPPORTED_VERSION:
            return supported_installed_versions[key]
    return ""

def _get_ansys_version_from_awp_root(awp_root):
    if awp_root.find("AWP_ROOT") >= 0:
        return int(awp_root.replace("AWP_ROOT", ""))

    return ""

def _get_sherlock_exe_path():
    ansys_base = _get_base_ansys()
    if not ansys_base:
        return ""
    if os.name == "nt":
        sherlock_bin = os.path.join(ansys_base, "sherlock", "SherlockClient.exe")
    else:
        **sherlock_bin = os.path.join(ansys_base, "sherlock", "runSherlock")**
    return sherlock_bin

### 📝 Steps to reproduce

Install pySherlock on any Linux box, and try to run launcher.launch_sherlock()

### 💻 Which operating system are you using?

Linux

### 📀 Which ANSYS version are you using?

Version 2025 R1 (25.1b20240718), build 20240718.2

### 🐍 Which Python version are you using?

3.10

### 📦 Installed packages

```shell
annotated-types==0.7.0
ansys-api-sherlock==0.1.29
-e git+https://github.com/ansys/pysherlock.git@b4ed83e11a3ecbf705466dd54521af217fc93da4#egg=ansys_sherlock_core
bleach==6.1.0
bokeh==3.3.4
Brotli==1.1.0
cachetools==5.3.2
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comtypes==1.3.1
contourpy==1.2.0
cycler==0.12.1
dill==0.3.8
fonttools==4.48.1
google-api-python-client==2.118.0
google-auth==2.27.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.62.0
grpcio==1.65.1
grpcio-tools==1.65.1
httplib2==0.22.0
idna==3.6
importlib-metadata==7.0.1
inflate64==1.0.0
Jinja2==3.1.3
keyboard==0.13.5
kiwisolver==1.4.5
linkify-it-py==2.0.3
Markdown==3.5.2
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.2
mdit-py-plugins==0.4.0
mdurl==0.1.2
multivolumefile==0.2.3
numpy==1.25.0
numpy-stl==3.1.1
packaging==23.2
pandas==2.1.4
panel==1.3.8
param==2.0.2
pillow==10.2.0
platformdirs==4.2.0
plotly==5.18.0
proto-plus==1.23.0
protobuf==3.20.3
psutil==5.9.7
py7zr==0.20.8
pyasn1==0.5.1
pyasn1-modules==0.3.0
PyAutoIt==0.6.5
pybcj==1.0.2
pycparser==2.21
pycryptodomex==3.19.1
pydantic==2.7.3
pydantic_core==2.18.4
pyparsing==3.1.1
pyppmd==1.1.0
python-dateutil==2.8.2
python-utils==3.8.2
pythonnet==3.0.3
pytz==2023.3.post1
pyviz_comms==3.0.1
pywin32==306
pywinauto==0.6.8
PyYAML==6.0.1
pyzstd==0.15.9
requests==2.31.0
rsa==4.9
scipy==1.12.0
selenium==3.141.0
six==1.16.0
tenacity==8.2.3
texttable==1.7.0
tornado==6.4
tqdm==4.66.2
typing_extensions==4.10.0
tzdata==2023.4
uc-micro-py==1.0.3
uritemplate==4.1.1
urllib3==1.26.16
vtk==9.3.0
webencodings==0.5.1
xyzservices==2023.10.1
zipp==3.17.0
```

anskhanson, 2024-07-26:
The read me has been updated with steps indicating the user must set the environment variables on Linux. Similar to PyMechanical.

aklara-ansys, 2024-07-26:
Thanks @anskhanson this is easy enough to add to my scripts

closed 2024-07-30 by ansnfernand