ansys/pysherlock #375: How to modify stencil thickness and solder type after importing odb++ file

NoahWeb24, 2024-09-09:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

After importing an ODB++ file using the project.import_odb_archive() function, I want to edit the circuit card properties, such as stencil thickness or default solder type. Currently, I am unsure how to access these properties. While the project.add_cca() function allows setting circuit card properties during the creation of a circuit card, it does not permit importing an ODB++ file into an existing circuit card. How can I access and edit the circuit card properties after importing an ODB++ file?

### 📝 Steps to reproduce

""

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024 R2

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
ansys-api-sherlock==0.1.31
ansys-sherlock-core==0.7.1
grpcio==1.66.1
importlib-metadata==4.13.0
protobuf==5.28.0
tk==0.1.0
zipp==3.20.1
```

ansjmoody, 2024-10-15:
The requested functionality is not available.
Changing this to an enhancement.

StiftMC, 2026-02-24:
Hi @ansjmoody!

This one is still relevant from our side. Maybe you can set a milestone for that. 

Thank you very much!

ansjmoody, 2026-09-14:
@ansnraines Sherlock needs a new API to update CCA properties. E.g. SherlockProjectService.updateCCA().

ansjmoody, 2026-10-05:
A Sherlock story has been created.