ansys/pysherlock #380: Error when updating laminate layer with certain resin percentages

NoahWeb24, 2024-09-16:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

When I use the `stackup.update_laminate_layer()` function to adjust the resin percentage in the glass construction list, I encounter an error message. However, when I manually adjust the resin percentage in the Sherlock program, it works without any issues. The code functions correctly for some predefined resin percentages, but fails for others.

### 📝 Steps to reproduce

from ansys.sherlock.core.launcher import launch_sherlock
sherlock = launch_sherlock()
sherlock.project.import_odb_archive(
    "ODB++ Tutorial.tgz",
    True,
    True,
    True,
    True,
    project="Test",
    cca_name="Card",
)
sherlock.stackup.update_laminate_layer(
    "Test",
    "Card",
    "2",
    "Generic",
    "FR-4",
    "Generic FR-4",
    0.015,
    "in",
    "106",
    [
        ("106", 68.0, 0.015, "in")
    ],
    "E-GLASS",
    "COPPER",
    "0.0",
)

Here's a revised version of your statement:

---

The example code provided by the API works correctly with a resin percentage of 68.0%. However, when I change the resin percentage to 50% for example, I encounter an error message.

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Sherlock 2024 R2

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
ansys-api-sherlock==0.1.31
ansys-sherlock-core==0.7.1
grpcio==1.66.1
grpcio-tools==1.66.1
importlib_metadata==8.4.0
protobuf==5.28.0
tk==0.1.0
zipp==3.20.1
```

ansjmoody, 2024-10-10:
I verified that the issue doesn't happen when the percentage is 68.0 and see the following error when changing the percentage to 50:
`ERROR - stackup - update_laminate_layer - Update laminate layer error: Invalid laminate style provided: 106`

ansjmoody, 2024-10-10:
This is happening because there is no construction with these properties:
style = 106
resin percentage = 50.0

When the style = 106, valid percentages are:
68.0, 69.0, 70.0, 71.0, 72.0, 73.0, 74.0, 75.0, 76.0, 77.0, 78.0

This can be seen in the UI when editing a laminate layer:
[image]

ansjmoody, 2024-10-15:
This issue has been fixed in Sherlock and is not a PySherlock issue. The fix will be included in Sherlock version 2025 R1.

closed 2024-10-15 by ansjmoody