ansys/pysherlock #407: Add API: delete_parts_from_parts_list()

NoahWeb24, 2024-10-10:
### 📝 Description of the feature

This function deletes a part or parts from the parts list of a specified project and CCA name. Additionally, it allows you to set a filter beforehand to delete multiple parts that match the filter criteria.

ansnraines, 2025-01-13:
@NoahWeb24 Would you have an example use case for this API request? Allowing filtering in the API significantly affects the scope of this work so trying to understand how you would apply this API.

NoahWeb24, 2025-01-14:
@ansnraines When navigating to the parts list, I can apply a filter from a text file located under 'Advanced Filter' using the load button. This filter enables me to display only the parts that are not included in the BOM. The API should support selecting this filter text file and automatically remove the parts that match the criteria specified in the file. Additionally, it would be helpful to have the possibility to provide the API with a list of reference designators to selectively remove specific parts from the parts list.

ansnraines, 2025-02-04:
@NoahWeb24 In architecting a solution, we'd like to propose two APIs: one for filtering and one for deleting based on Ref Des. For the filtering API, it would be parts_list_filter() with part properties as optional parameters (instead of importing a text file). Would this satisfy your requirements?

closed 2025-04-21 by ansjmoody