ansys/pysherlock #414: Bug located in ...error in sherlock launcher

namdev420, 2024-10-14:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

ERROR - launcher - launch_sherlock - <_InactiveRpcError of RPC that terminated with:
        status = StatusCode.UNIMPLEMENTED
        details = "Method not found: SherlockCommonService/isSherlockClientLoading"
        debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"Method not found: SherlockCommonService/isSherlockClientLoading", grpc_status:12, created_time:"2024-10-14T16:48:11.2034919+00:00"}"

### 📝 Steps to reproduce

from ansys.sherlock.core.launcher import launch_sherlock
sherlock = launch_sherlock()

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Ansys sherlock 2022 R2

### 🐍 Which Python version are you using?

3.11

### 📦 Installed packages

```shell
PS C:\Users\E0721133\pysherlock> python -m pip freeze
absl-py==2.1.0
altgraph==0.17.4
ansys-api-sherlock==0.1.34
-e git+https://github.com/ansys/pysherlock.git@d50044941319722aeb02dc4f07731f91f7dc4714#egg=ansys_sherlock_core
anyio==4.4.0
argon2-cffi==23.1.0
argon2-cffi-bindings==21.2.0
arrow==1.3.0
asgiref==3.8.1
asttokens==2.4.1
astunparse==1.6.3
async-lru==2.0.4
attrs==24.2.0
babel==2.16.0
beautifulsoup4==4.12.3
bleach==6.1.0
blinker==1.8.2
certifi==2024.7.4
cffi==1.17.0
charset-normalizer==3.3.2
click==8.1.7
cloudpickle==3.0.0
colorama==0.4.6
comm==0.2.2
contourpy==1.2.1
cx_Freeze==7.2.0
cx_Logging==3.2.0
cycler==0.12.1
dash==2.17.1
dash-core-components==2.0.0
dash-html-components==2.0.0
dash-table==5.0.0
debugpy==1.8.5
decorator==5.1.1
defusedxml==0.7.1
devtools==0.12.2
Django==5.1
et-xmlfile==1.1.0
executing==2.0.1
fastjsonschema==2.20.0
filelock==3.15.4
Flask==3.0.3
flatbuffers==24.3.25
fonttools==4.53.1
fqdn==1.5.1
fsspec==2024.6.1
gast==0.6.0
google-pasta==0.2.0
grpcio==1.65.4
h11==0.14.0
h5py==3.11.0
httpcore==1.0.5
httpx==0.27.0
idna==3.7
image2pdf==1.0.0
importlib_metadata==8.2.0
iniconfig==2.0.0
ipykernel==6.29.5
ipython==8.26.0
isoduration==20.11.0
itsdangerous==2.2.0
jedi==0.19.1
Jinja2==3.1.4
joblib==1.4.2
json5==0.9.25
jsonpointer==3.0.0
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
jupyter-events==0.10.0
jupyter-lsp==2.2.5
jupyter_client==8.6.2
jupyter_core==5.7.2
jupyter_server==2.14.2
jupyter_server_terminals==0.5.3
jupyterlab==4.2.4
jupyterlab_pygments==0.3.0
jupyterlab_server==2.27.3
keras==3.4.1
kiwisolver==1.4.5
libclang==18.1.1
lief==0.14.1
Markdown==3.6
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.9.1.post1
matplotlib-inline==0.1.7
mdurl==0.1.2
mistune==3.0.2
ml-dtypes==0.4.0
MouseInfo==0.1.3
mpmath==1.3.0
namex==0.0.8
nbclient==0.10.0
nbconvert==7.16.4
nbformat==5.10.4
nest-asyncio==1.6.0
networkx==3.3
notebook_shim==0.2.4
numpy==1.26.4
opencv-python==4.10.0.84
openpyxl==3.1.5
opt-einsum==3.3.0
optree==0.12.1
outcome==1.3.0.post0
overrides==7.7.0
packaging==24.1
pandas==2.2.2
pandocfilters==1.5.1
parso==0.8.4
pdf2image==1.17.0
pdfminer==20191125
pefile==2023.2.7
pillow==10.4.0
platformdirs==4.2.2
plotly==5.23.0
pluggy==1.5.0
prometheus_client==0.20.0
prompt_toolkit==3.0.47
protobuf==4.25.4
psutil==6.0.0
pure_eval==0.2.3
PyAutoGUI==0.9.54
pycparser==2.22
pycryptodome==3.20.0
PyGetWindow==0.0.9
Pygments==2.18.0
pyinstaller==6.10.0
pyinstaller-hooks-contrib==2024.8
PyMsgBox==1.0.9
PyMuPDF==1.24.9
PyMuPDFb==1.24.9
pyparsing==3.1.2
PyPDF2==3.0.1
pyperclip==1.9.0
PyQt5==5.15.11
PyQt5-Qt5==5.15.2
PyQt5_sip==12.15.0
PyRect==0.2.0
PyScreeze==0.1.30
PySide6==6.7.2
PySide6_Addons==6.7.2
PySide6_Essentials==6.7.2
PySocks==1.7.1
pytest==8.3.2
python-dateutil==2.9.0.post0
python-json-logger==2.0.7
pytweening==1.2.0
pytz==2024.1
PyVirtualDisplay==3.0
pywin32==306
pywin32-ctypes==0.2.2
pywinpty==2.0.13
PyYAML==6.0.2
pyzmq==26.1.0
qt5-applications==5.15.2.2.3
qt5-tools==5.15.2.1.3
referencing==0.35.1
requests==2.32.3
retrying==1.3.4
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rich==13.7.1
rpds-py==0.20.0
scikit-learn==1.5.1
scipy==1.14.0
seaborn==0.13.2
selenium==4.23.1
Send2Trash==1.8.3
shiboken6==6.7.2
six==1.16.0
sniffio==1.3.1
sortedcontainers==2.4.0
soupsieve==2.5
spyder-kernels==2.5.2
sqlparse==0.5.1
stack-data==0.6.3
sympy==1.13.1
tenacity==9.0.0
tensorboard==2.17.0
tensorboard-data-server==0.7.2
tensorflow==2.17.0
tensorflow-intel==2.17.0
tensorflow-io-gcs-filesystem==0.31.0
termcolor==2.4.0
terminado==0.18.1
threadpoolctl==3.5.0
tinycss2==1.3.0
torch==2.4.0
tornado==6.4.1
traitlets==5.14.3
trio==0.26.2
trio-websocket==0.11.1
types-python-dateutil==2.9.0.20240316
typing_extensions==4.12.2
tzdata==2024.1
uri-template==1.3.0
urllib3==2.2.2
wcwidth==0.2.13
webcolors==24.8.0
webencodings==0.5.1
websocket-client==1.8.0
Werkzeug==3.0.3
wrapt==1.16.0
wsproto==1.2.0
xlwings==0.31.10
zipfile39==0.0.8.0
zipp==3.19.2
```

ansjmoody, 2024-10-15:
PySherlock is not compatible with Sherlock 2022 R2.

namdev420, 2024-10-16:
which version of Pysherlock is compatible with sherlock 2022 R2?

ansjmoody, 2024-10-16:
There aren't any versions of PySherlock that are compatible with Sherlock 2022 R2.
PySherlock is compatible with Sherlock 2023 R2 or later. PySherlock is backwards compatible, so installing the latest should work.
Note that some APIs may not be available for the version of Sherlock that is installed. The error in this bug is an example of what happens when an API is not available. This will be improved when versioning is implemented.

closed 2024-10-16 by ansjmoody