ansys/pysherlock #439: launch_sherlock(single_project_path="") doesnt open project

NoahWeb24, 2024-11-05:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

When I try to open a specific project using single_project_path, the project does not load—instead, only the main Sherlock application opens. The Sherlock Client Console returns the following error:
java.lang.Exception: Missing Sherlock project folder -singleProject E:\ProjectPath\MyProject

After this, Sherlock defaults to loading the standard path:
Loading C:\Users\mayingen\AppData\Roaming\Sherlock\packageImages

Upon investigating the launcher.py code, I found that the args list being passed to subprocess.Popen(args) is incorrectly formatted. Here’s what the args list looks like when running the launch_sherlock() function:
['C:\\Program Files\\ANSYS Inc\\v242\\sherlock\\SherlockClient.exe', '-grpcPort=9090', '-singleProject "E:\ProjectPath\MyProject"']

However, for the command to work correctly, -singleProject and the project path must be separate entries in the list, like this:
['C:\\Program Files\\ANSYS Inc\\v242\\sherlock\\SherlockClient.exe', '-grpcPort=9090', '-singleProject', 'E:\ProjectPath\MyProject']

To work properly, the -singleProject flag and the project path need to be separate arguments. At least, that works for me.
Or did I make a mistake in the string for single_project_path?

### 📝 Steps to reproduce

from ansys.sherlock.core.launcher import launch_sherlock
launch_sherlock(single_project_path="Path to your Project")

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Sherlock 2024R2

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

ansys-api-sherlock==0.1.31
ansys-dpf-core==0.9.0
ansys-dpf-gate==0.4.1
ansys-dpf-gatebin==0.4.1
ansys-dpf-post==0.5.0
ansys-grpc-dpf==0.8.1
ansys-sherlock-core==0.7.1
cachetools==5.5.0
certifi==2024.8.30
charset-normalizer==3.4.0
colorama==0.4.6
customtkinter==5.2.2
darkdetect==0.8.0
et-xmlfile==1.1.0
google-api-core==2.22.0
google-api-python-client==2.149.0
google-auth==2.35.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.65.0
grpcio==1.66.1
grpcio-tools==1.66.1
httplib2==0.22.0
idna==3.10
importlib_metadata==8.4.0
numpy==1.24.4
openpyxl==3.1.5
packaging==24.1
pandas==2.0.3
pillow==10.4.0
proto-plus==1.25.0
protobuf==5.28.0
psutil==6.1.0
pyasn1==0.6.1
pyasn1_modules==0.4.1
pyparsing==3.1.4
python-dateutil==2.9.0.post0
pytz==2024.2
pywin32==308
requests==2.32.3
rsa==4.9
scooby==0.10.0
six==1.16.0
tk==0.1.0
tqdm==4.66.6
tzdata==2024.1
uritemplate==4.1.1
urllib3==2.2.3
zipp==3.20.1

kinielsen, 2024-11-26:
Hi Team, 

I am experiencing the same problem. 

@NoahWeb24 Can your please tell me how you corrected the launcher.py file, in order for this to work? 

Best regards, 
Kim

ansjmoody, 2025-01-17:
This is fixed in pysherlock version 0.8.

closed 2025-01-17 by ansjmoody