ansys/pysherlock #453: launcher.py

aklara-ansys, 2024-11-19:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Unable to use '-noGUI' arguments or even any command arguments with the _**sherlock_cmd_args**_ argument in launcher.launch_sherlock() function.

### 📝 Steps to reproduce

>>> launcher.launch_sherlock(sherlock_cmd_args="-noGUI")
['C:\\WB_INSTALL\\v252\\sherlock\\SherlockClient.exe', '-grpcPort=9090', "['-noGUI']"]

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

11:12:05.699 INFO: Version 2025 R1 (25.1), build 20241111.1

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
C:\Users\aklara>python -m pip freeze
annotated-types==0.7.0
ansys-api-sherlock==0.1.31
ansys-sherlock-core==0.7.1
bleach==6.1.0
bokeh==3.3.4
Brotli==1.1.0
build==1.2.1
cachetools==5.3.2
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comtypes==1.3.1
contourpy==1.2.0
cycler==0.12.1
dill==0.3.8
exceptiongroup==1.2.2
fonttools==4.48.1
gitdb==4.0.11
GitPython==3.1.43
google-api-python-client==2.118.0
google-auth==2.27.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.62.0
grpcio==1.66.1
grpcio-tools==1.65.1
httplib2==0.22.0
idna==3.6
importlib-metadata==7.0.1
inflate64==1.0.0
iniconfig==2.0.0
Jinja2==3.1.3
keyboard==0.13.5
kiwisolver==1.4.5
linkify-it-py==2.0.3
Markdown==3.5.2
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.2
mdit-py-plugins==0.4.0
mdurl==0.1.2
multivolumefile==0.2.3
numpy==1.25.0
numpy-stl==3.1.1
packaging==23.2
pandas==2.1.4
panel==1.3.8
param==2.0.2
pillow==10.2.0
platformdirs==4.2.0
plotly==5.18.0
pluggy==1.5.0
proto-plus==1.23.0
protobuf==5.28.3
psutil==5.9.7
py7zr==0.20.8
pyasn1==0.5.1
pyasn1-modules==0.3.0
PyAutoIt==0.6.5
pybcj==1.0.2
pycparser==2.21
pycryptodomex==3.19.1
pydantic==2.9.2
pydantic_core==2.23.4
pyparsing==3.1.1
pyppmd==1.1.0
pyproject_hooks==1.1.0
pytest==8.3.3
python-dateutil==2.8.2
python-utils==3.8.2
pythonnet==3.0.4
pytz==2023.3.post1
pyviz_comms==3.0.1
pywin32==306
pywinauto==0.6.8
PyYAML==6.0.1
pyzstd==0.15.9
requests==2.31.0
rsa==4.9
scipy==1.12.0
selenium==3.141.0
six==1.16.0
smmap==5.0.1
tenacity==8.2.3
texttable==1.7.0
tomli==2.0.1
tornado==6.4
tqdm==4.66.2
typing_extensions==4.10.0
tzdata==2023.4
uc-micro-py==1.0.3
uritemplate==4.1.1
urllib3==1.26.16
vtk==9.3.0
webencodings==0.5.1
xyzservices==2023.10.1
zipp==3.17.0
```

closed 2025-01-20 by ansjmoody

ansjmoody, 2025-01-20:
This is fixed in PySherlock 0.8.2.

aklara-ansys, 2025-01-23:
Verified this is working in pySherlock 0.9.0 

>>> launcher.launch_sherlock(sherlock_command_args="-noGUI")
INFO - launcher - launch_sherlock - Command arguments: ['C:\\WB_INSTALL\\v252\\sherlock\\SherlockClient.exe', '-grpcPort=9090', '-noGUI']
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
ERROR - common - check - Health check failed.
INFO - common - check - Connection is healthy.
INFO - common - check - Connection is healthy.
INFO - common - is_sherlock_client_loading - Sherlock client has finished initializing.
<ansys.sherlock.core.sherlock.Sherlock object at 0x0000028A31CC1030>
>>>

PR #487 (2025-01-20): fix: changed Launcher.launch_sherlock() to properly append sherlock_command_args
## Description
Modify Launcher.launch_sherlock() to add sherlock_command_args to the list of arguments. It was appending the string representation of the list to the arguments.

## Issue linked
Closes #453

## Checklist:
- [] Run unit tests and make sure they all pass
		- Run tests without Sherlock running
		- Run tests with Sherlock GRPC connection
- [] Check and fix style errors
		- pre-commit command line check
		- Problems tab in PyCharm
- [] Bench test new/modified APIs by using and modifying the code in the  [... cut]