ansys/pysherlock #472: UpdateNaturalFrequencyPropsRequest API

kinielsen, 2024-12-10:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

Dear Team, 

I found a bug when updating Natural Frequency Analysis Properties. 

UpdateNaturalFrequencyPropsRequest API accepts "float" in SherlockAnalysisService_pb2.pyi

[image]

It results in Float number when updating Natural Frequency Properties. 

[image]

Best regards, 
Kim 

### 📝 Steps to reproduce

sherlock = launch_sherlock(port = 9090)

sherlock.analysis.update_natural_frequency_props(
          project=project_name,
          cca_name=pcb_name,
          natural_freq_count=6,
          natural_freq_min=20,
          natural_freq_min_units="HZ",
          natural_freq_max=2000,
          natural_freq_max_units="HZ",
          part_validation_enabled=False,
          require_material_assignment_enabled= False,
          analysis_temp=22.0,
          analysis_temp_units="C"
          )

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

2024R2

### 🐍 Which Python version are you using?

3.12

### 📦 Installed packages

```shell
annotated-types==0.7.0
ansys-api-sherlock==0.1.35
ansys-sherlock-core==0.8.1
greenlet==3.1.1
grpcio==1.67.1
grpcio-tools==1.67.1
jproperties==2.1.2
numpy==2.1.3
pandas==2.2.3
pika==1.3.2
protobuf==5.29.0
psycopg2==2.9.10
pydantic==2.10.2
pydantic_core==2.27.1
python-dateutil==2.9.0.post0
pytz==2024.2
PyYAML==6.0.2
setuptools==75.1.0
six==1.16.0
SQLAlchemy==2.0.36
typing_extensions==4.12.2
tzdata==2024.2
wheel==0.44.0
```

kinielsen, 2024-12-11:
Hi again, 

I have tried to change the naturalFreqMin and naturalFreqMax in UpdateNaturalFrequencyPropsRequest to "bulletin.int" but with no success. 

Also, it is not possible to leave out any parameters in the function as you will get an "Analysis.update_natural_frequency_props() missing X required positional arguments:" error. 

Best regards, 
Kim

ansjmoody, 2025-01-20:
@kinielsen These values are defined as `float` in the .proto file, but an integer value is required by Sherlock. That is the reason for the wording of the comments.
In your example, you are providing integer values, which should update the values to what is specified.
I would expect an error returned in the response if you change the example to use a floating point value for either of these fields.

Did you enter this because Sherlock is showing ".0" in the values?

I.e. Sherlock shows this:
Min Frequency: 20.0
Max Frequency: 2000.0

Instead of:
Min Frequency: 20
Max Frequency: 2000

ansjmoody, 2025-04-14:
Sherlock 25.2 fixes the issue with showing ".0" for Min and Max Frequency fields.

@kinielsen Do not modify the .pyi files. These are generated files that are built using the .proto file definitions.

ansjmoody, 2025-04-15:
@kinielsen Regarding not being able to leave out parameters, that is because this API is designed to require all fields. Can you clarify what you'd like to happen in the case where not all of the parameters are provided?

closed 2025-04-28 by ansjmoody