ansys/pysherlock #524: sherlock.core.parts.get_parts_list_properties not returning all properties

aklara-ansys, 2025-03-14:
### 🔍 Before submitting the issue

- [x] I have searched among the existing issues
- [x] I am using a Python virtual environment

### 🐞 Description of the bug

get_parts_list_properties() is not returning pad properties for parts that have pad fields enabled. 

[image]

### 📝 Steps to reproduce

To reproduce, import the Tutorial Project.zip from the v252/sherlock/tutorial folder.

request = GetPartsListPropertiesRequest(project="Tutorial Project", cca_name="Main Board", ref_des=["R53"]
test = sherlock.parts.get_parts_list_properties(request)
for i in range(len(test[0].properties)):
	print(test[0].properties[i].name, test[0].properties[i].value)

### 💻 Which operating system are you using?

Windows

### 📀 Which ANSYS version are you using?

Version 2025 R2 (25.2b20250311), build 20250311.1

### 🐍 Which Python version are you using?

3.9

### 📦 Installed packages

```shell
annotated-types==0.7.0
ansys-additive-core==0.18.2
ansys-api-additive==1.7.2
ansys-api-platform-instancemanagement==1.1.0
ansys-api-sherlock==0.1.35
ansys-platform-instancemanagement==1.1.2
ansys-sherlock-core==0.8.2
bleach==6.1.0
bokeh==3.3.4
Brotli==1.1.0
build==1.2.1
cachetools==5.3.2
certifi==2024.2.2
cffi==1.16.0
charset-normalizer==3.3.2
clr-loader==0.2.6
colorama==0.4.6
comtypes==1.3.1
contourpy==1.2.0
cycler==0.12.1
dill==0.3.8
exceptiongroup==1.2.2
fonttools==4.48.1
gitdb==4.0.11
GitPython==3.1.43
google-api-core==2.24.0
google-api-python-client==2.118.0
google-auth==2.27.0
google-auth-httplib2==0.2.0
googleapis-common-protos==1.62.0
grpcio==1.66.1
grpcio-health-checking==1.62.3
grpcio-tools==1.65.1
httplib2==0.22.0
idna==3.6
importlib-metadata==7.0.1
inflate64==1.0.0
iniconfig==2.0.0
Jinja2==3.1.3
keyboard==0.13.5
kiwisolver==1.4.5
linkify-it-py==2.0.3
Markdown==3.5.2
markdown-it-py==3.0.0
MarkupSafe==2.1.5
matplotlib==3.8.2
mdit-py-plugins==0.4.0
mdurl==0.1.2
multivolumefile==0.2.3
numpy==1.25.0
numpy-stl==3.1.1
packaging==23.2
pandas==2.1.4
panel==1.3.8
param==2.0.2
pillow==10.2.0
platformdirs==4.2.0
plotly==5.18.0
pluggy==1.5.0
proto-plus==1.23.0
protobuf==5.29.3
psutil==5.9.7
py7zr==0.20.8
pyasn1==0.5.1
pyasn1-modules==0.3.0
PyAutoIt==0.6.5
pybcj==1.0.2
pycparser==2.21
pycryptodomex==3.19.1
pydantic==2.9.2
pydantic_core==2.23.4
pyparsing==3.1.1
pyppmd==1.1.0
pyproject_hooks==1.1.0
pytest==8.3.3
python-dateutil==2.8.2
python-utils==3.8.2
pythonnet==3.0.4
pytz==2023.3.post1
pyviz_comms==3.0.1
pywin32==306
pywinauto==0.6.8
PyYAML==6.0.1
pyzstd==0.15.9
requests==2.31.0
rsa==4.9
scipy==1.12.0
selenium==3.141.0
six==1.16.0
smmap==5.0.1
tenacity==8.2.3
texttable==1.7.0
tomli==2.0.1
tornado==6.4
tqdm==4.66.2
typing_extensions==4.10.0
tzdata==2023.4
uc-micro-py==1.0.3
uritemplate==4.1.1
urllib3==1.26.16
vtk==9.3.0
webencodings==0.5.1
xyzservices==2023.10.1
zipp==3.17.0
```

ansjmoody, 2025-04-14:
This was a Sherlock bug, not pysherlock.
I verified that it does not happen in Sherlock version 20250411.1.

aklara-ansys, 2025-04-24:
@ansjmoody this still happens for me only in pySherlock - Sherlock Version 2025 R2 (25.2b20250422), build 20250422.1. The BloomRPC returned the pad properties, but I don't see them among the list of get_parts_list_properties:

[image]

aklara-ansys, 2025-04-29:
Closing this issue, I do see the pad properties displayed in the latest build

[image]